echostack-org/echoshader #3: Get sample data for GSoC

lsetiawan, 2022-03-10:
Need small datasets for GSoC get started from the https://osoceanacoustics.github.io/echopype-examples/ms_OOI_EK60_mooringtimeseries.html and https://osoceanacoustics.github.io/echopype-examples/echopype_tour.html

valentina-s, 2022-03-12:
I also have a [short notebook](https://colab.research.google.com/drive/1PJtvoY5rqL6ocFaNp2BdMcN_16Vywk0I#scrollTo=hNnLAbs00F2m.) showing how to read and plot an echogram with a few .nc files:
I could PR it with the dataset if you all think it would be useful.

leewujung, 2022-03-13:
@valentina-s : Please do! I think that will be very useful.

One thing (which does not require immediate action), that in [echopype #578](https://github.com/OSOceanAcoustics/echopype/pull/579) we are changing the name `range` to `echo_range` and `water_level` to `range_offset`. As the ones @lsetiawan will be adding, we just need to remember to change the notebooks and update the datasets once the next echopype release is out.

leewujung, 2022-03-14:
From conversation with @lsetiawan : we also need the raw data, because the platform location (ship track) is in there.

emiliom, 2022-03-14:
> we also need the raw data, because the platform location (ship track) is in there.

I think it'd be preferable if the raw data provided are *converted* raw data

leewujung, 2022-03-14:
Yes absolutely @emiliom -- that is what I meant!

leewujung, 2023-04-10:
Recap on latest status from 2022:
- @valentina-s added a notebook under `./notebooks` that accesses and plots a long nc file: [this notebook](https://github.com/leewujung/echoshader/blob/main/notebooks/Read_multiple_sonar_nc_files_zip.ipynb)
- @ldr426 included data pulling sections in notebooks under [./docs/source/gsoc/2022](https://github.com/OSOceanAcoustics/echoshader/tree/main/docs/source/gsoc/2022)

For this year, we'll create dedicated notebooks or scrips for pulling large volumes of data and keep them in s3 bucket for developing viz access/plotting/optimization. I've created #70 to keep track of this, and will close this issue now.

closed 2023-04-10 by leewujung