echostack-org/echoshader #5: Add getting started section to understand/manipuate the data structure

leewujung, 2022-03-10:
We need a section in the README to help potential contributors understand what the data contains, such as:
- what are in the calibrated `Sv` xr dataset, such as the dimensions
- links to resources for netCDF files and zarr format
- tips/resources to use libraries such as `xarray` to access/manipulate the data
- etc

lsetiawan, 2022-03-10:
Some resources:
- [xarray tutorial](https://xarray-contrib.github.io/xarray-tutorial/)
- [zarr](https://zarr.readthedocs.io/en/stable/)
- [netcdf python](https://unidata.github.io/netcdf4-python/)

lsetiawan, 2022-03-10:
Great get started resource on holoviz straight from the creator himself: https://nbviewer.org/github/philippjfr/pydata-2021/blob/master/PyData_2021.ipynb

lsetiawan, 2022-03-10:
Okay one more cool site, awesome panel: https://awesome-panel.org/

ldr426, 2022-03-11:
Nice to meet you! My name is Dingrui Lei, a GSoC candidate. It is very nice to list these newbie-friendly guides which help understand data structure we will deal with. Thank you!

lsetiawan, 2022-03-11:
Hi @ldr426 It's nice to meet you as well. I'm Don, one of the mentor. I'm glad you are finding these guides helpful. We are currently in the process of consolidating these together into our getting started guide. So more useful documents to come! Stay tuned. 😄

leewujung, 2022-03-14:
Closing this now that we can move to more refined items in the getting started doc.