echostack-org/echoshader #6: Add issues to capture the desired types of visualization

leewujung, 2022-03-10:
@b-reyes @lsetiawan @valentina-s @emiliom
Based on the discussion we just had, let's create issues to capture the type of visualizations we'd like to see for this project.

We also need to create a label for this so it is easier to follow. 
What should be the label name? viz_wanted"? is there another name that is more general that is easily recognizable for what this project wants? `gsoc2022_wanted`?

lsetiawan, 2022-03-10:
I vote for `gsoc2022_wanted`

lsetiawan, 2022-03-10:
Some past discussion on pangeo brainstorming:
- Demonstrate data-reduction and scalable visualization tools/approaches, such as datashader and other HoloViz components
- For ship data, extract/preprocess the ship track information to generate readily usable spatial information. Turn transect lat-lon points into simple transect lines in a GeoJSON, which can then be clickable. Click on a transect line on a map figure, and have a plot figure below show the depth vs distance echogram/Sv data
- Something like the below figure for OOI data, each slice is data from one day; similar to the above, users can select which day (or maybe a week?) they want to see through a drop-down menu or slidebar

[image]
**ASL's Data Cube Representation of Time Series Illustrating Seasonal Variations**

lsetiawan, 2022-03-10:
Some other plotting things that we can get ideas from? 🤷🏽‍♂️ : https://csiro-acoustics.github.io/Visualize-IMOS-Bioacoustics-data/

leewujung, 2022-03-10:
Yep, the cube thing was an echopype issue https://github.com/OSOceanAcoustics/echopype/issues/79  :D 
Now we can close that and have it here!

lsetiawan, 2022-03-10:
With interactive 3D I have a feeling we will need to end up using plotly to make this happen, so participants might need to expand on holoviews plotly support! http://holoviews.org/gallery/index.html

Which would be AMAZING IMO! 😄

ldr426, 2022-03-11:
I have a idea for issues79. When I read the guide of xarray, I found it can set "gorupby" = "time.day"/"time.week" to add a slidebar near the hvplot picture, which users can use to see what's happening at a specific time. I tried it, and it can work, which means I can see the pictures at any time period from start_date to end_date. Have you considered this?

lsetiawan, 2022-03-11:
Hey @ldr426! Thanks for this idea. Do you think you can [open an issue](https://github.com/OSOceanAcoustics/echoshader/issues/new?assignees=&labels=gsoc+ideas+2022&template=gsoc-ideas.md&title=) about this? It would be great to capture this in an issue so that everyone can chime in there. Please provide any screenshots and/or code that you've tried! Would love to check it out. Thanks! 😄

leewujung, 2022-03-11:
I second @lsetiawan 's suggestion! Welcome @ldr426 !

leewujung, 2023-04-10:
The issues were created so this can be closed now -- a year too late!  😛

closed 2023-04-10 by leewujung