echostack-org/echoshader #10: A possible idea for through a slidebar

ldr426, 2022-03-12:
## Questions

> https://github.com/OSOceanAcoustics/echoshader/issues/6#issuecomment-1064600486

 For DataArray whose time Coordinate called "time", we can set groupby="time.day" or groupby="time.week" to see the situation at certain time. 
[image]
But for the MVBS, the time Coordinate in it is "ping_time". It can't set groupby="ping_time.day". That will report errors. And setting groupby="ping_time.day" doesn't make sense.
[image]
 In the above picture, I just set x-axis="frequency" to show it can work, leave it. If we want to show one day or one week's MSBC, we can add a Coordinate called "time_range" to help set a cycle we want to show. "time_range" can represent a day or a week. Its values are consistent with "ping_time". 
`mvbs_plot = ds_MVBS.sel(frequency=200000).hvplot.image(
    x='ping_time',y='depth',**groupby='time_range',**
    color='Sv', 
    cmap='jet', clim=(-80, -30)
).options(invert_yaxis=True)`
That may help add a sidebar to select the date.I haven't yet finished all guides, maybe it can do direct operation on "ping_time".

ldr426, 2022-03-12:
I will look for some solutions in holoviews too.

ldr426, 2022-03-13:
If using hvplot to draw xarray image diagram, it can only choose two coordinates for x-axis and y-axis respectively. But other coordinates that are not selected for the diagram will automatically add according sidebars. (If we pick a different, lower dimensional plot type (such as a ‘line’) it will automatically apply a groupby over the remaining dimensions.) And it cannot choose time coordinate both as x-axis and slidebar option at the same time like xarray line diagram, probably because dynamic loading will cause conflicts. So if we wanna show different days' Sv, we can add one more coordinate on behalf with the day or the week we want to check.

ldr426, 2022-03-13:
I have not yet studied Panel. It will combine the map and Sv diagram, I think.

leewujung, 2022-03-16:
> It can help combine the map and Sv diagram

#18  😄

ldr426, 2022-03-16:
> I think

After studying panel, I start to know there is no need to change Sv data structure.

closed 2022-03-17 by ldr426