echostack-org/echoshader #12: Add concrete getting started steps

leewujung, 2022-03-13:
From @b-reyes 's work and @valentina-s 's comments in #5, let's add concrete steps to help folks get started.

Suggestions:
- read example files with xarray (#11 is one example, and those in [echopype examples](https://github.com/OSOceanAcoustics/echopype-examples))
- make a widget which displays the sonar Sv variable with ping_time on x coordinate and range_bin on y coordinate, and allow the user to change the frequency, and the colormap.
- share examples in matplotlib plots so that contributors can try to convert them to interactive plots

lsetiawan, 2022-06-15:
@leewujung can this be closed now? Seems like the GSoC participant is pretty clear on how to get started at this point? Thanks.

leewujung, 2023-04-10:
Closing this -- almost a year later 😆

We have notebooks from @ldr426 now as the new starting points. In the next couple weeks I'll add a few things to keep track of for this year.

closed 2023-04-10 by leewujung