echostack-org/echoshader #17: Visualize sonar data as a stack of cards

leewujung, 2022-03-14:
For sonar data from a stationary platform in the ocean (often called a "mooring"), we can see movements of animals up and down the water column according to the daylight cycle. 

The data cube representation basically plots 1 image for each day, from this type of sonar time series, and arrive at the following figure:
[image]
**Data Cube Representation of Time Series Illustrating Seasonal Variations**
(Image from ASL Environmental Sciences https://aslenv.com/AZFP-data.html)

It will be cool to have this in echoshader!  
We can have widgets to interactively control the time span, which frequency to plot, colormap, etc.

ldr426, 2022-03-15:
https://user-images.githubusercontent.com/56751303/158410279-679950d7-6779-4f23-be23-13623f6af59e.mp4

Hi, mentors. It is not hard to control frequency or colormap by using panel, but it may get conflict when setting time as x-axis and controlling time-span at the same time. I am trying to solve this problem.

ldr426, 2022-03-15:
[image]
Is time range-slider better? Or normal slider to choose fixed time span like a day?

leewujung, 2022-03-16:
@ldr426 : cool! I am thinking the beginning and ending **date** is fixed depending on the input data, but then where the open face is "slices out" depends on a single date, so it seems that a normal slider would work. Thoughts?

In terms of controlling the x-axis (assuming in the context of this data cube), the front face should always have the same x-axis since it is always a 24-hr period, and y-axis be along the same depth, but the y-z plane would have different length depending on the start/slice/end dates.

ldr426, 2022-03-16:
@leewujung : got it! How about using DatePicker? It can fix start/end date, then limit the dates that can be selected. As shown below, I set start/end date according to ping_time input. It is much more convenient to use DatePicker than normal slider, I think.
[image]

ldr426, 2022-03-17:
@leewujung wow! 
Does plotting the **top face** mean we only considerring "the top", like depth=0m or -5m, near the sea surface?  
The new component can use scatter plot, whose y-axis is date and 24-hour is x-axis. If Sv values > -75, then can be put on the plot. Or line plot better?

leewujung, 2022-03-17:
> Does plotting the top face mean we only consider "the top", like depth=0m or -5m, near the sea surface?

Actually this is interesting -- this will be cool thing, to visualize how the DVM sandglass shape at different depth! So then you can slice the original cube either along depth like what you said, and also slice it along date, like the original thought.

> The new component can use scatter plot, whose y-axis is date and x-axis is 24-hour. If Sv values > -75, then can be put on the plot. Or line plot better?

Scatter plot is an interesting idea: What if for Sv > -75 (or some other threshold) you scale the size of the scatter points with the Sv value? Might would be good to use something like `xr.coarsen` to reduce the number of points if it gets crowded.

I wonder if a scatterplot and an image may give different insight to the data!

ldr426, 2022-03-17:
@leewujung Thank you for answering.
yeah, it would be more cool to scale the size of points according to values. 
Now it is much more interesting and challenging. I will try it! :smiley:

ldr426, 2022-03-17:
@leewujung  already finished datepicker, frequency and colormap part.
I will do more work for Diel vertical migration (DVM) and design a more beautiful interface before pulling request for this issue.
ps: I just use two consecutive hours(23-24,0-1) from two consecutive days as data input, so x-axis doesn't show 24-hour. Also some data before 24:00 in raw data are vacant. When using enough big and complete data, it will show a whole day.

https://user-images.githubusercontent.com/56751303/158801192-a3e4c972-3be6-4480-a4db-1d63fc4c4cf7.mp4

ldr426, 2022-03-17:
@leewujung 
should we design a template page just like this. put echoshader logo and offer more friendly user-interface. That would be much cooler than just showing it in jupyter cell, I think
[image]

leewujung, 2022-03-21:
> some data before 24:00 in raw data are vacant

Yes, that is true. The instrument had a small off period before midnight each day.

closed 2023-08-10 by ldr426