echostack-org/echoshader #20: Plot statistics from sonar data interactively

leewujung, 2022-03-14:
Our first visualization of sonar data is usually a 2D image that shows how the echoes change across time and space, like the one shown in our [Contributor's guide](https://github.com/OSOceanAcoustics/echoshader/blob/main/gsoc_contributors_guide.md).

But once we want to analyze the data more, it will help to pull statistics out from the data.

For example, we can have in one panel a small pannable box on an echogram, and then in another panel have the histogram of Sv values contained within that window. Something like below:

[image]

There are other statistics that can be pulled out and very useful, like the distribution of the Sv differences across frequency, the shape of the Sv distribution summarized as skewness and kurtosis, etc, that can be added to this to make the visualization more informative.

ldr426, 2022-03-19:
this part can integrate into https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-116866906.It can be designed in same panel.
We can do some statistics on cross section, too, like analyzing the difference or trend in Sv at the everyday same time.

leewujung, 2022-03-21:
> We can do some statistics on cross section, too, like analyzing the difference or trend in Sv at the same everyday-time.

Sounds great!

> this part can integrate into https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-116866906. It can be designed in the same panel.

Absolutely. So, the central idea of this project is to create "a bag of widgets" in the package that allow people to mix and match different components in a panel dashboard. Do you wanna think about how that design might look like, and open up an issue (an idea) for that?

ldr426, 2022-03-22:
@leewujung 
> Absolutely. So, the central idea of this project is to create "a bag of widgets" in the package that allow people to mix and match different components in a panel dashboard. Do you wanna think about how that design might look like, and open up an issue (an idea) for that?

OK, mentor Lee :blush:. I just did a prototype of echoshader GUI, https://github.com/OSOceanAcoustics/echoshader/issues/27#issue-1176397969. Please give me more feedback on layout.

ldr426, 2022-03-23:
I want to ask that what x and y coordinates of this histogram represent?
[image]

ldr426, 2022-03-23:
>For example, we can have in one panel a small pannable box on an echogram, and then in another panel have the histogram of Sv values contained within that window. Something like below:

How about using wheel zoom tool instead of pannable box. Users can drag and zoom with mouse. It has better operation and efficiency.

https://user-images.githubusercontent.com/56751303/159684149-1ad0b2f6-0150-4638-a8e1-6c1a2cf68057.mp4

leewujung, 2022-03-23:
Can we do both? Since a lot of times users want to know how things look like while keeping view of the bigger context (pannable box), but yes they may want to zoom in also (drag/zoom).

ldr426, 2022-03-24:
> know how things look like while keeping view of the bigger context (pannable box)

nice hint. :laughing:  I almost forgot that we need keeping view of the bigger context to make this sense.

emiliom, 2022-03-29:
> What is skewness and kurtosis? please give me a link about these mathematic terms blush Then I will try to plot and link.

They are statistical parameters. Here are some references:
- https://community.gooddata.com/metrics-and-maql-kb-articles-43/normality-testing-skewness-and-kurtosis-241
- https://en.wikipedia.org/wiki/Skewness
- https://en.wikipedia.org/wiki/Kurtosis

ldr426, 2022-03-30:
Thanks for answering, Mr. @emiliom
Now I undestand skewness and kurtosis are parameters representing normality of plots. 
But I am still not clear what _histogram_ should be plotted from the box selected on an echogram.
>then in another panel have the histogram of Sv values contained within that window. 

What is the _histogram of Sv values_? If user selects a box from echogram, it would be a small 3d image(x: ping_time, y: depth, colotmap value: Sv value). _histogram of Sv values_  's y-axis, I think, should represent Sv value. What does x-axis represent for, depth or ping_time? If it is ping_time, we should caculated mean Sv values for different depths.

emiliom, 2022-03-30:
>> then in another panel have the histogram of Sv values contained within that window.
>
> What is the histogram of Sv values? If user selects a box from echogram, it would be a small 3d image(x: ping_time, y: depth, colotmap value: Sv value). histogram of Sv values 's y-axis, I guess, should represent Sv value. What does x-axis represent for, depth or ping_time? If it is ping_time, we should caculated mean Sv values for different depths.

I think what @leewujung meant was a histogram of Sv values in the selected echogram box. In other words:
- x axis: Sv value bins for each histogram bar (say, -90 to -80 dB, -80 to -70 dB, etc)
- y axis: cell counts or normalized distribution (bin cell count / total cell count, as 0 - 1)
- the depth and ping_time axis "disappear", since we're now treating the box of values as a simple collection of values

ldr426, 2022-03-31:
Ms @leewujung, please give me some examples about what you said "the distribution of the Sv differences across frequency".
How to plot that? Overlay or layout histograms of Sv for different kinds of frequency?
And below is a demo for histogram of Sv. 

https://user-images.githubusercontent.com/56751303/161028038-f1d554c7-e69d-429f-ab7b-e635e1cf420a.mp4

leewujung, 2022-04-15:
@ldr426 : somehow missed this message and just saw this now!

Not sure what if this is what you and @bitanb1999 mean by a layout histogram, but i was envisioning having the x axis of the histogram of each frequency channel to be identical with these panels stacked vertically (like sharey=True for matplotlib subplots). This way people can see where the peak and spread of the Sv values look like for multiple frequency channels at once.

ldr426, 2022-04-15:
Ms @leewujung:  I got it :) And there are two options.
#### What I mean about layout histograms 
is to plot more than one subplot. Just like 'sharey=True for matplotlib subplots', _different plots_  are stacked vertically or horizontally. Something like below.
[image]

#### And 'overlay' 
is to draw these layouts in the _same plot_.  Something like below. Different colors represent for different frequencies. 
That will make the contrast stand out more. 
[image]

leewujung, 2022-04-15:
Oops I realized that I had a typo. I meant share**x**=True instead of y.

I think both would be fine, and we can even make it a toggle option, just for the overlay one we need to more careful in selecting the colors.

bitanb1999, 2022-04-16:
@leewujung Understood! Also, @ldr426 , didn't realise the usage of the overlay histogram before. Thanks for making it clear!

closed 2023-08-10 by ldr426