echostack-org/echoshader #26: 3D echogram curtain along ship track

leewujung, 2022-03-15:
For sonar data collected from moving platforms (say, a ship), the echogram can be thought of as a sheet of curtain draped along the GPS track of the ship. 

This should be really fun with some other components, like linking the aerial view of the ship track in 2D as in #18.

I couldn't find a good example so I made a sketch:
[image]

The closest example I found that does the same thing just the 3D perspective is more from the top (birds'-eye view) is this one:
[image]
From [Kang et al. 2019. Fig. 2](https://www.researchgate.net/publication/333721623_Fishes_distribution_and_their_connection_to_artificial_reefs_off_Bukchon_Jeju_Island_using_geographic_information_system). The legend is reproduced here:
> Fig. 2. Artificial reefs with the echogram curtain (a) and reefs and fish schools with the echogram curtain (b). The sphere size is the reef volume and the sphere color means the reef material (a). The sphere color is the Sv of fish school detected and the sphere size means the fish school area (b). The halo (semi-transparent circle) indicates the number of reef installed. The black layer is the echogram curtain (a and b).

ldr426, 2022-03-25:
Hi, mentors. There is a question for this issue :smiley:

Folium is an excellent map library for https://github.com/OSOceanAcoustics/echoshader/issues/18, providing flexible and pannble maps in jupyter. 
It is much better to be deployed and operated than GeoPandas or cartopy. 
[image]

But it can't be achieved to plot 3D view directly on map, no matter GeoPandas or Folium. We can plot 3D echogram curtain in a new interactive panel, something like as follows. 

Users can view various angles and zoom in/out by mouse. That's enough if we just want to observe the shape. Except for shape of the curtain, is there anything more should be conveyed through this plot?

https://user-images.githubusercontent.com/56751303/160122400-8a1154ef-dc02-4b73-912b-e88faaa30040.mp4

leewujung, 2022-03-26:
@ldr426 : The wiggly shape of the curtain is informative as the ship track, but the more important is often the variation of where strong scattering features are (e.g., dense fish school) above a certain bathymetry features (e.g., near [shelf break](http://www.coastalwiki.org/wiki/File:OceanZones.jpg)). Not sure if I understood your question correctly?

What is that caldera-looking feature? looks nice! What library/tools did you use to plot this? 😀

ldr426, 2022-03-26:
@leewujung  Thank you for replying, mentor Lee :)
>What is that caldera-looking feature? looks nice! What library/tools did you use to plot this? 😀

The caldera-looking plot is an example from library [PyVista](https://docs.pyvista.org/index.html). It is good for us to built 3D plotting. 

>the more important is often the variation of where strong scattering features are

I am confused how to refelct the variation of where strong scattering features are. 
Does that mean offering information about curtain(geoinfo) and Sv value at the same time?  
If so, My idea is to plot Sv on curtains by using colormap, like something below. :point_down:
[image]

valentina-s, 2022-03-26:
@ldr426 Good find on the PyVista? Can you superimpose a bathymetry surface together with the curtain. If you use this  colormap which is specific for echograms the patterns will show naturally:
```python
# install echopype from a PR to use the ek500 colormap
!pip install -q git+git://github.com/OSOceanAcoustics/echopype.git@6af147cc0a2e2665c52091f0e8f3ac7c7e6e6601
import echopype
cm_ek500 = matplotlib.cm.get_cmap('ep.ek500')
Sv.isel(frequency=2).resample(ping_time='30s').mean().T.plot(yincrease=False, vmax=-34, vmin=-76, cmap=cm_ek500)
```

ldr426, 2022-03-26:
@valentina-s    Hi, mentor Staneva 😀
>Can you superimpose a bathymetry surface together with the curtain?

I think it can be achieved when bathymetry files with corresponding lat&lon offered. 
mentor @lsetiawan, could you give me some bathymetry files to let me try to plot bathymetry surface first?

emiliom, 2022-03-26:
Hi @ldr426 ! It looks like PyVista is a wrapper interface to VTK. I would suggest exploring the VTK panel elements, since they are already integrated with the holoviz ecosystem. I haven't used them myself, but they look promising. There are VTK panel demos in the [Panel gallery](https://panel.holoviz.org/gallery/index.html). See for example the [VTKslicer demo](https://panel.holoviz.org/gallery/demos/VTKSlicer.html#demos-gallery-vtkslicer)

> Folium is an excellent map library for https://github.com/OSOceanAcoustics/echoshader/issues/18, providing flexible and pannble maps in jupyter.
> It is much better to be deployed and operated than GeoPandas or cartopy.

I just wanted to point out that comparing Folium or cartopy to GeoPandas is a bit of an apples and oranges comparison. GeoPandas is more about the underlying geospatial data structure and data operations. It does provide some basic, convenient plotting capability, but you need specialized libraries like cartopy and Folium (and [ipyleaflet](https://ipyleaflet.readthedocs.io/en/latest/) to produce terrific, interactive visualizations). All the geospatial visualization libraries support data in GeoPandas GeoDataframes. I could envision a folium or ipyleaflet interactive map element that lets users select the geospatial area or segments, then VTK panes are used to generate the 3D curtain plots for the selected areas!

ldr426, 2022-03-27:
Thanks for suggestions, Mr. @emiliom :)
>I would suggest exploring the VTK panel elements, since they are already integrated with the holoviz ecosystem. 

That is very useful! It inspires me to create a panel widget which interconnects 2D map(folium/leaflet) and 3D topographic map. 

The challenge now is no matter VTK or PyVista, there is little interoperability directly with xarray. 
see https://github.com/pydata/xarray/issues/4470#issue-710357592.

So that will be novel and fun to do some job like constructing a 3D PyVista/VTK plot representation of a DataArray or createing unstructured grids from DataArrays.

leewujung, 2022-03-27:
> plot Sv on curtains by using colormap

Yes, that was what we have in mind also, as a first shot for 3D viz over bathymetry. 😄  Once the framework is there it would be straightforward to swap out the content of the curtain. For example one can show some Sv-derived statistics over some coarser grids (coarser than the original Sv resolution).

> No matter VTK or PyVista, there is little interoperability directly with xarray. see https://github.com/pydata/xarray/issues/4470#issue-710357592.
So that will be novel and fun to do some job like createing unstructured grids from DataArrays and ploting and constructing a 3D PyVista/VTK representation of a DataArray with correct size.

This is a very interesting and relevant thread!
Thinking through what the Sv data dimension and coordinates are and what is needed in this use case:
```
Sv(frequency, ping_time, range_bin)  # calibrated output from compute_Sv
```
with an additional coordinate variable:
```
range(frequency, ping_time, range_bin)  # this can be converted to depth in meters
```
To attach the lat/lon position to the Sv data, we'll need to interpolate (using `ping_time`) the lat/lon position from the `Platform` group in the converted files.
We then have something like
```
Sv(frequency, ping_time, range_bin)
lat(ping_time)
lon(ping_time)
```
Let's assume on the simpler case that the mapping from `range_bin` (units: digitization sample number) to `range` (units: meters, distance of echo from the sonar instrument)  is completely identical for all `frequency` channels. `range` can subsequently be converted to `depth` (units: meters, actual depth in the ocean), then this becomes
```
Sv(frequency, ping_time, depth)
lat(ping_time)
lon(ping_time)
```
Or a fully unstructured form
```
Sv(frequency, ping_time)
depth(ping_time)
lat(ping_time)
lon(ping_time)
```

leewujung, 2022-03-27:
@ldr426 : forgot point out one small thing, that the depth axis is inverted in your plots in this thread, but no the one with the mooring (#17). This is because the OOI sonar was mounted on the seafloor looking up, and the ship sonar was mounted on the ship looking down. 😀

ldr426, 2022-03-28:
Thanks for your useful explanation! Ms lee @leewujung :)
The explanation makes it crystal clear for understading the data structure and helps me organize my idea that how to create the API doing some standardized data transformation and integration.

leewujung, 2022-03-28:
@ldr426 : for data standardization we can leverage echopype functionality (and of course create the "linkage" when needed). I encourage you to discuss with us as you sketch those steps out, as we are continuing to develop echopype and other tools that may come into play in parallel. @emiliom and @lsetiawan have a lot of experience in data standardization and workflows. 😎

closed 2023-08-10 by ldr426