echostack-org/echoshader #27: A prototype of echoshader GUI

ldr426, 2022-03-22:
## Ideas
It is a prototype of echoshader GUI.
[image]

### Main area 
This area is main area of the application. It plots and offers description.
[image]

There is a tab for different needs.
Overview:https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-1168669060
Analyze: https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-1168669060 and https://github.com/OSOceanAcoustics/echoshader/issues/20
Geospatial data:https://github.com/OSOceanAcoustics/echoshader/issues/18#issue-1168692706 and https://github.com/OSOceanAcoustics/echoshader/issues/26#issue-1169037627
[image]

### Sidebar
Users can configure different variables in this area, like which date or color map to select. We can add more customization later, like line width or window type for histogram.
[image]

This logo is actually a hyperlink. Users can click it to visit echoshader's official website.
[image]

ldr426, 2022-03-23:
Or main area can be divided into three or more sub-areas

leewujung, 2022-03-23:
Thanks @ldr426! 
I think for echoshader our goal is to create the elements ("widgets") that a user can select and put together their own "customizable GUI" (or a better word is probably "dashboard" in Panel language). 

So, it does go with what you have laid out above in terms of what can go where, but we want users to be able to customize it based on their own needs, and we don't decide that for them. Or, they could simply use these widgets in a Jupyter notebook (i.e. outside of a Panel dashboard). 

Does this make sense?

ldr426, 2022-03-24:
@leewujung
Got it :smile:. Users can call related API to plot what they want in Jupyter notebook. Sometimes, even combined charts. So we don't need an integrated panel dashboard.
There is no essential difference between showing widgets in Jupyter and Panel dashboard. But Panel dashboard is good for server deployment and prettier layout.

leewujung, 2022-03-24:
Yes! And I should say that there's nothing wrong with you laying out a dashboard though. That would be really nice demo of the capability!

ldr426, 2022-03-24:
@leewujung  
Thanks, mentor Lee!
It is also a very nice training for me to learn and use more about Panel. :runner:

lsetiawan, 2022-06-15:
This idea is superseded by the idea of creating widgets rather than full GUI for users. Closing this as stale.

closed 2022-06-15 by lsetiawan