echostack-org/echoshader #29: Plot 3D sonar data using Pyvista

ldr426, 2022-04-02:
## Ideas
I used to think using scatter to plot the top face of the data cube in https://github.com/OSOceanAcoustics/echoshader/issues/17#issuecomment-1070342858
But actually compared to 3D plotting, visual effect of scatter is not straightforward and clear.
So I try plotting this in Pyvista. the effect is nice. So maybe we can use this to plot data cube surface.

[image]
[image]

leewujung, 2022-04-09:
This is interesting. I do agree that plotting a surface/image would be more efficient than using scatter -- I thought you used hvImage for the implementation you showed in #17 though? Maybe I misunderstood?

> https://github.com/OSOceanAcoustics/echoshader/issues/17#issuecomment-1070342858 can be divided into two parts. One is a stack of cards implemented by holoviews and panel, another is a data cube giving brid view(actually an overview from all angles), implemented by pyVista.

I like this idea! This way the view won't be over complicated and very flexible. 🤓

ldr426, 2022-04-09:
>I thought you used hvImage for the implementation you showed in https://github.com/OSOceanAcoustics/echoshader/issues/17 though?

My implementation in https://github.com/OSOceanAcoustics/echoshader/issues/17 is using hvImage to implement cross section view(x:time, y: depth), but not yet bird view(x:time in a day, y: date). To implement bird view in hvImage needs some data transformation. Because it cannot simply set up x=ping_time.time and y=ping_time.date, which does works in scatter, but not image. But bird view in image is what should be achieved in the official work period. 

By the way, I have submitted my proposal on GSoC. If there are any problems or advice, please tell me, Ms @leewujung. :grinning:

bitanb1999, 2022-04-13:
Hello, @leewujung I was also thinking of the 3D plots that can be made using ideas such as this article https://towardsdatascience.com/visualizing-three-dimensional-data-heatmaps-contours-and-3d-plots-with-python-bd718d1b42b4 instead of PyVista as to reduce the dependency on other packages.

leewujung, 2022-04-15:
Hey @ldr426 and @bitanb1999 ! I think both of these are good ideas and it may just come down to efficiency and the type of information one wants from the data.

When you have something in the form of (X, Y, Z) where Z is some function of X and Y, whether it is plotted as a color-coded surface like the ones @ldr426 showed or a color-coded 3D “relief” @bitanb1999 referred to comes down to which one conveys the type of information in the data. 3D viz is cool to play with and can reveal interesting trends that are not seen in a 2D color plot since our eyes respond to the height variation (Z) very well. But 3D views depending on the viewing angle some “landscape” features can block some

ldr426, 2022-04-15:
Ms @leewujung 
>once the data gets bigger the performance would suffer

When data gets bigger, coarsening data is very necessary. Just set up dataset.points[::100] or even dataset.points[::1000] .When zooming out, resolution will be higher for a coarse data cube. 

There are many other ways to improve performance for big data cube. For example, developers can only plot the six faces for the cube, if users' main concern is what cube surface looks like.

leewujung, 2022-04-15:
@ldr426 : I think only subsample data in the way you described can lead to loss of features. For example, if you have some spiky signal that comes in every 100 samples, and you do [5::100] then that would be lost. We can take a look at what datashader does and explore different algorithms to see what works better for echosounder data. We can also compute mean Sv as a very simplistic way to reduce the size of data to be plotted.

ldr426, 2022-04-15:
Ms @leewujung:  Thanks for mentioning loss of some features like spiky signal. I ignored it before.
>We can take a look at what datashader does

Method rasterize() in holoviews uses Datashader to realize big data visualization. It renders every data into a histogram, where every array cell counts data points falling into that pixel. Then each cell turns into a pixel in the image. very interesting! :))

bitanb1999, 2022-04-16:
> 

Sounds like a goodway to start with. Can I use this as a GSoC proposal idea?

bitanb1999, 2022-04-16:
> Ms @leewujung
>
>
> When data gets bigger, coarsening data is very necessary. Just set up dataset.points[::100] or even dataset.points[::1000] .When zooming out, resolution will be higher for a coarse [...]

I believe this will make us lose out on important information and features as Ms @leewujung has stated. Also, some sort of PCA would be useful when visualising things in datacubes to choose the features wisely and prevent feature loss optimally.

bitanb1999, 2022-04-16:
> Hey @ldr426 and @bitanb1999 ! I think both of these are good ideas and it may just come down to efficiency and the type of information one wants from the data.
> 
> When you have something in the form of (X, Y, Z) where Z is some function of X and Y, whether it is plotted as a color-coded surface like the ones @ldr426 showed or a color-coded 3D “relief” @bitanb1999 referred to comes down to which one conveys the type of information in the data. 3D viz is cool to play with and can reveal interesting trends that are not seen in a 2D color plot since our eyes respond to the height variation (Z) very well. But 3D views depending on the viewing angle some “landscape” features can be blocked sometimes, so it is often more suitable for interactive visualization.
> 
> On the efficiency front, we do think that once the data gets bigger the performance would suffer. For example, if we want to visualize a month of data from the OOI or the hake survey, things will stall. We’ll need to benchmark these with small and big datasets in the initial phase to determine which way to go.

Can the benchmarking pipeline be used as a GSoC proposal?

ldr426, 2022-04-16:
>some sort of PCA would be useful when visualising things in datacubes
@bitanb1999 , Bitan, I think we do not deal with a scatter point cloud. Actually the data cube is made of a stack of 'cards'. Maybe PCA is not applicable for this.

bitanb1999, 2022-04-17:
> > some sort of PCA would be useful when visualising things in datacubes
> 
> @bitanb1999 , Bitan, I think we do not deal with a scatter point cloud. Actually the data cube is made of a stack of 'cards'. Maybe PCA is not applicable for this.

Aah makes sense!

leewujung, 2022-04-18:
> Can the benchmarking pipeline be used as a GSoC proposal?

Yes, we encourage applicants to include benchmarking using small and large test datasets in their proposals!

bitanb1999, 2022-04-18:
> > Can the benchmarking pipeline be used as a GSoC proposal?
> 
> Yes, we encourage applicants to include benchmarking using small and large test datasets in their proposals!

Also, @leewujung could you please review my proposal before the final submission?

leewujung, 2022-04-18:
@ldr426 :

> Method rasterize() in holoviews uses Datashader to realize big data visualization. It renders every data into a histogram, where every array cell counts data points falling into that pixel. Then each cell turns into a pixel in the image. It is a good solution for overplotting. I think the problem about cube is not overplotting. It's that there are too many 'pixels'. :))

Not sure what you meant by "overplotting," but isn't plotting too many pixels than what's needed (which also needs to be defined) indeed a form of "overplotting" ?

leewujung, 2022-04-18:
@ldr426 @bitanb1999 -- Thanks for submitting your proposals! We'll review your proposals and provide one round of suggestions via email tomorrow (Apr 18 US time) early evening.

bitanb1999, 2022-04-18:
> @ldr426 @bitanb1999 -- Thanks for submitting your proposals! We'll review your proposals and provide one round of suggestions via email tomorrow (Apr 18 US time) early evening.

Thank you!

ldr426, 2022-04-18:
Ms @leewujung:
Occlusion of data by other data is called [overplotting](https://anaconda.org/jbednar/plotting_pitfalls/notebook) or overdrawing. There are too many points falling in the same situation(pixel). something like this below.
[image]

Datashader just highlight the color of each pixel(situaiton) to solve this problem.
Below are datashader pictures. The more highlighted parts suffer from more overplotting. 
One point of Sv datacube only occupy one pixel. But there are too many points. 
So I said "cube is not overplotting. It's that there are too many 'pixels'. :))"

[image]

[image]

leewujung, 2022-04-18:
@ldr426 : great, I did not know that these specific terms but am familiar with the problems. I think the confusion above was from what a "pixel" means.

leewujung, 2022-04-18:
>> some sort of PCA would be useful when visualising things in datacubes

> I think we do not deal with a scatter point cloud. Actually the data cube is made of a stack of 'cards'. Maybe PCA is not applicable for this.

PCA could be done if you reshape the data -- see this paper by @valentina-s and myself: [Compact representation of temporal processes in echosounder time series via matrix decomposition](https://asa.scitation.org/doi/10.1121/10.0002670). We used NMF but showed PCA in a supplemental figure. 😄

bitanb1999, 2022-04-18:
> PCA could be done if you reshape the data -- see this paper by @valentina-s and myself: [Compact representation of temporal processes in echosounder time series via matrix [...]

Thank you for the paper. Will go through this as I believe PCA will help us in 3D plotting without losing important features!

ldr426, 2022-04-18:
Ms @leewujung 
interesting! Seems like that there is no traditional matrix factorization methods that can handle spatiotemporal autocorrelation features. It is my first time to hear about NMF. It is a great way to accurately mine spatiotemporal patterns in massive travel flows. very interesting new knowledge 😄 ! When it comes to matrix factorization, SVD(Singular Value Decomposition) could do some work for compressing the cubes. 

But after some thinking, I'm wondering if we're going in the wrong direction. Maybe we should create a lightweight and flexible data visualization tool. If there is a cube, what users only can see is just the six surfaces. But users should have access to the inside of the cube! My new idea is to create a control component to **customize the length, width and height of the cube , but only draw six surfaces each time**. That will more much much more efficient and flexible! What do you think, Ms lee?

valentina-s, 2022-04-18:
@ldr426 @bitanb1999 One way to visualize what is inside the cube is through volume ray tracing. This is often used in biomedical imaging and graphics.  It also can be computational expensive for large datasets, but GPU acceleration can help. Often libraries such as [OpenGL](https://www.opengl.org/), [WebGL](https://get.webgl.org/), [Three.js](https://threejs.org/) are used, but they are not particularly friendly to scientists. There is a python wrapper [ipyvolume](https://ipyvolume.readthedocs.io/en/latest/examples/volshow.html).

bitanb1999, 2022-04-26:
> @ldr426 @bitanb1999 One way to visualize what is inside the cube is through volume ray tracing. This is often used in biomedical imaging and graphics. It also can be computational expensive for [...]

https://dl.acm.org/doi/pdf/10.1145/147130.147155 Just went through the process of Volume Ray tracing and through this particular research paper that proposes the fastest volume ray tracing algo.

leewujung, 2023-05-21:
I'll close this issue to declutter, but I created a new label "good reference" so that we can find useful issues containing lots of information later easily.

closed 2023-05-21 by leewujung