echostack-org/echoshader #30: Map location_time to ping_time

ldr426, 2022-04-04:
Ms @leewujung  mentioned below in https://github.com/OSOceanAcoustics/echoshader/issues/26#issue-1169037627
>Thinking through what the Sv data dimension and coordinates are and what is needed in this use case:
Sv(frequency, ping_time, range_bin)  # calibrated output from compute_Sv
with an additional coordinate variable:
`range(frequency, ping_time, range_bin)  # this can be converted to depth in meters`
To attach the lat/lon position to the Sv data, we'll need to interpolate (using ping_time) the lat/lon position from the Platform group in the converted raw files.
We then have something like
`Sv(frequency, ping_time, range_bin)`
`lat(ping_time)`
`lon(ping_time)`

But for lat/lon position from the Platform group, the time coordinate is not 'ping_time', but 'location_time'. 
The number of location_time is about three times as number of ping_time. 
So what I do to map from location_time to ping_time is taking one every few 'location_time's . In other words, reducing location_time's number to equal ping_time's number so that it could subsitute ping_time in Sv dataArrays in correct dimension.
```
def avg_adjacent(location_list,ping_time_num):
    res = []  # compressed results
    interval_num = len(location_list) // ping_time_num
    for i in range(ping_time_num):
        res.append(location_list[i*interval_num])
    return res
```

ldr426, 2022-04-05:
Sometimes the EchoData.platform. sentence_type is not uniform. It can cause the interweaving pattern like below.
In this situation, we should unified interpreted sentence to smooth the line.
[image]

leewujung, 2022-04-05:
> But for lat/lon position from the Platform group, the time coordinate is not 'ping_time', but 'location_time'.

Yes this is why I mentioned that we need to interpolate (using ping_time) the lat/lon position from the Platform group.

> The number of location_time is about three times as number of ping_time.
> So what I do to map from location_time to ping_time is taking one every few 'location_time's . In other words, reducing [...]

You can take a look at xarray [reindex_like](https://xarray.pydata.org/en/stable/generated/xarray.DataArray.reindex_like.html) which gives you options for the operation you implemented as a loop.

Two hints that may be useful to think about:
- the final dimension and time points should be identical to the acoustic (Sv) data
- it is useful to think about the data as vectors and matrices, so that you don't have to loop over a matrix to get every single element. Libraries like pandas and xarray has many built-in vector operations and it usually makes the code a lot cleaner and faster. Matlab is another language that you may have seen that is predominantly vector/matrix-based.

leewujung, 2023-05-21:
I'll close this issue now since in later versions of echopype we've implemented functions to add lat/lon locations to the Sv dataset, and names of the time coordinates have been updated to follow the convention.

closed 2023-05-21 by leewujung