echostack-org/echoshader #34: Design the APIs for use as widgets

ldr426, 2022-04-20:
## Ideas
Ms @leewujung once reminded me to include some thoughts in my proposal on how I would design the API for use as widgets. But the day when Mr. Lee gave this advice is already the deadline for GSoC application. I don't have enought time to add the design patterns in the proposal. I will list some thoughts there in the next few days.

ldr426, 2022-04-20:
### For visualizing sonar data as a stack of cards in https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-1168669060

- **Component 1** : [DatetimeRangePicker](https://panel.holoviz.org/reference/widgets/DatetimeRangePicker.html#widgets-gallery-datetimerangepicker).  Control the time span.
[image]

- **Component 2** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Choose specific frequency and colormap.
[image]

- **Component 3** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Custom colormap input.
[image]

ldr426, 2022-04-20:
### For visualizing sonar data as a data cube in https://github.com/OSOceanAcoustics/echoshader/issues/17#issue-1168669060

- **Component 1** : [RangeSlider](https://panel.holoviz.org/reference/widgets/RangeSlider.html#widgets-gallery-rangeslider) :  Customize width and height of the cube. 
Width : Selected date
Height : Depth
[image]

- **Component 2** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Choose specific frequency and colormap.
[image]

- **Component 3** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Custom colormap input.
[image]

ldr426, 2022-04-21:
### For plotting statistics from sonar data interactively in https://github.com/OSOceanAcoustics/echoshader/issues/20#issue-1168728349
- **Component 1** : [DatetimeRangePicker](https://panel.holoviz.org/reference/widgets/DatetimeRangePicker.html#widgets-gallery-datetimerangepicker).  Control the time span.

- **Component 2** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Choose specific frequency and colormap.

- **Component 3** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Customize colormap input.

- **Component 4** : [Toggle](https://panel.holoviz.org/reference/widgets/Toggle.html).
Toggle between showing single frequency plots and all frequencies plots.
Toggle between layout mode and out laymode (see in https://github.com/OSOceanAcoustics/echoshader/issues/20#issuecomment-1100275045).

- **Component 5** : [TextEditor](https://panel.holoviz.org/reference/widgets/TextEditor.html#widgets-gallery-texteditor) Output information like skewness and kurtosis.
<image src="https://user-images.githubusercontent.com/56751303/164531513-e5140bbb-fc6b-4810-a82b-1ab2e607193a.png" width=500px>

- **Editable box**: Four selection modes: Overwrite; Intersect; Union; Inverse
<image src="https://user-images.githubusercontent.com/56751303/164532013-11123b18-5ff8-4577-944a-9a4a5d1fc649.png" width=500px>

### Prototype 
made by balsamiq
[image]

ldr426, 2022-04-22:
### For Interactive visualization of sonar data and geospatial location in https://github.com/OSOceanAcoustics/echoshader/issues/18#issue-1168692706

- **Component 1** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Choose specific frequency and colormap and tiles.
Tiles: Square bitmap graphics displayed in a certain kind of grid arrangement to show a map.
[image]

- **Component 2** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Customize colormap input.

- **Component 3** : [TextEditor](https://panel.holoviz.org/reference/widgets/TextEditor.html#widgets-gallery-texteditor). Output geo information like latitude, longitude and time **in selected box**

- **Selected box**:  Select a box in map plot, and show the corresponding Sv image and geo info.

### Prototype 
made by balsamiq
[image]

ldr426, 2022-04-23:
### For 3D echogram curtain along ship track in https://github.com/OSOceanAcoustics/echoshader/issues/26#issue-1169037627
This part can link the aerial view of the ship track in 2D and map with editable box.
We can use editable box in echogram and map to specify the area of this 3D curtain.

- **Component 1** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Choose specific frequency, colormap.

- **Component 2** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Customize colormap input.

- **Component 3** : [TextEditor](https://panel.holoviz.org/reference/widgets/TextEditor.html#widgets-gallery-texteditor). Output geo information like latitude, longitude and time **in a selected box**

- **Component 4** : [Toggle](https://panel.holoviz.org/reference/widgets/Toggle.html).
To do some interesting extends! :smile:
Toggle between showing bathymetry and not 
(see in https://github.com/OSOceanAcoustics/echoshader/issues/26#issuecomment-1079644854)
Toggle between showing underwater vehicle track and not 
(see in https://github.com/OSOceanAcoustics/echoshader/issues/28#issuecomment-1088282857
 and https://github.com/OSOceanAcoustics/echoshader/issues/28#issuecomment-1088570395)

### Prototype 
made by balsamiq
[image]

ldr426, 2022-04-23:
Ms @leewujung : Just done the design patterns which are not included in the proposal.
These are just sketchy prototypes to demonstrate core functionality. 
I will add more components to do some styling work( like add css in html 😏).
Please tell me what components or functions you want to add in these prototypes. 😄
It would be wonderful to create a comprehensive and detailed prototype before coding begins!

leewujung, 2022-04-24:
Hey @ldr426 -- thanks for putting together these! I think these are great that it helps one think very clearly on what are the parameters needed for each form of the visual displays.

By API I was actually referring to the programmatic design of how we structure the echoshader library and how users would interact with it. But after thinking about this some more, I think this part is better done via in-depth discussions between contributors and the mentor team. We can continue to brainstorm here, but for sure we as a team need to talk more once the program actually starts.

There are a few things that we should think about:
- What are the common set of components that we can specify, probably using [param](https://param.holoviz.org/)?
- How/where do we define the parameters/components interaction with the underlying echodata (like slicing a particular frequency)? When does computation get executed?
- Where do computation related to data sit and get executed?
  - My initial thoughts are that the data-intensive computation part would be handled by echopype, and echoshader only deals with the visualization component. This still seems a good choice to me, but we should think this through. For example, combining multiple EchoData objects or data files should be executed in echopype, but it is probably natural for users to want to point the source to a folder with whole bunch of files and wanting to display them. 
    - for display only involving calibrated echo data (`FILENAME_Sv.nc/zarr` files), this in theory would just take a 1 liner xarray `open_mfdataset`, but we need to make it error-proof and test performance with different sizes of datasets.
    - for display involving lat/lon, because those data are NOT part of the `FILENAME_Sv.nc/zarr` files and are in the converted raw files (`FILENAME.nc/zarr`), we'd need to interpolate the lat/lon to the echo data grid. I think @ldr426 you asked about this in one of the issues. This should be in echopype since it is a general functionality, but we haven't built that yet. The question here though is when does this get executed, and do the user get a copy of the interpolated data back if those do not already exist? Or do we force the users to generate such a dataset (with both echo and lat/lon) first and then feed it into echoshader? @lsetiawan I think you mentioned this in one of our discussions.
- In terms of structuring the library, between [Declarative Panels](https://panel.holoviz.org/getting_started/#declarative-panels), [widgets and params](https://panel.holoviz.org/user_guide/Param.html), and [templates](https://panel.holoviz.org/user_guide/Templates.html), where do we want each of our visualization options to sit?
- once we decide and build these, it should be straightforward to put in notebooks in the docs to explain usage and our design philosophy and choices we made.

Haha, the above has become a sprawling comment. We can move them to a separate issue at some point, once we have more thoughts flushed out.

leewujung, 2023-05-21:
This is another good reference issue that I'll be closing now to declutter. But I've added the "good reference" label so that we can find this easily.

closed 2023-05-21 by leewujung