echostack-org/echoshader #36: New ability to explore statistics of Sv

lsetiawan, 2022-06-15:
## Overview

Echoshader needs the ability to plot statistics from the Sv data so that users can see where the peak and spread of the Sv values for multiple channels/frequencies.

### Expected Results

Users can choose an editable box on an echogram in one panel, and then in another panel
have the histogram of Sv values contained within that window. Users can also check the
distribution of the Sv differences across frequency and the shape of the Sv distribution
summarized as skewness and kurtosis.

<details>
<summary><h3>Proposed idea</h3></summary>
<br>

### For plotting statistics from sonar data interactively in https://github.com/OSOceanAcoustics/echoshader/issues/20#issue-1168728349
- **Component 1** : [DatetimeRangePicker](https://panel.holoviz.org/reference/widgets/DatetimeRangePicker.html#widgets-gallery-datetimerangepicker).  Control the time span.

- **Component 2** : [Select](https://panel.holoviz.org/reference/widgets/Select.html). Select specific frequency and colormap.

- **Component 3** : [MultiSelect](https://panel.holoviz.org/reference/widgets/MultiSelect.html#widgets-gallery-multiselect). Select **multiple frequency values** to show combined plots.
- **Component 4** : [ArrayInput](https://panel.holoviz.org/reference/widgets/ArrayInput.html#widgets-gallery-arrayinput). Customize colormap input.

- **Component 5** : [Toggle](https://panel.holoviz.org/reference/widgets/Toggle.html).
Toggle between layout mode and outlay mode (see in https://github.com/OSOceanAcoustics/echoshader/issues/20#issuecomment-1100275045).

- **Component 6** : [TextEditor](https://panel.holoviz.org/reference/widgets/TextEditor.html#widgets-gallery-texteditor). Output hist information like skewness and kurtosis.

- **Editable box in Sv image**: Four selection modes: Overwrite; Intersect; Union; Inverse
<image src="https://user-images.githubusercontent.com/56751303/164532013-11123b18-5ff8-4577-944a-9a4a5d1fc649.png" width=500px>

### Prototype 
made by balsamiq
[image]

_Originally posted by @ldr426 in https://github.com/OSOceanAcoustics/echoshader/issues/34#issuecomment-1105633269_
</details>

## Actions

@ldr426 please create small issues on the various steps that you're going to do to make this happen so that we can track it here. Thank you. Let me know if you have any questions.

ldr426, 2022-06-16:
[image]
Until now, I still use hvPlot to plot xarray data, because it offer exact interface to input xarray directly.  Holoviews can also deal with xarray data, which needs some transformation, sometimes can be onerous. Holoviews are primarily based on bokeh, a low-level tool which is based on Javascript. But A low-level tool is highly configurable, with an expressive but relatively verbose command language that makes it possible to precisely control how it works. 
For example, getting data in box select cannot be done by hvPlot, which should be processed in holoview or even bokeh.
hvPlot, Holoviews and boken's relationship is like below. 
[image]
So in initial work, I have to read lots of documents to get familiar with these python viz libraries. So I can know which tool is the most suitable for a specific task. [Here](https://pyviz.org/tools.html#high-level-shared-api) lists some most common python viz libraries. If only using high-level tools, some accurate and subtle work would never be done. So it maybe burdensome to learn libraries related to holoview. 
[image]

leewujung, 2023-05-21:
I'll close this now since most of the capabilities required here already exist, and we'll focus on refactoring the package structure and improving/benchmarking the performance going forward.

Just added the "good reference" label so that we can check back here for info later.

closed 2023-05-21 by leewujung