echostack-org/echoshader #51: 3D curtain in Plotly

ldr426, 2022-07-29:
There are two ways to create a 3D curtain in Plotly.

- use Surface
Every single surface comprises a ribbon. Create ribbons for every depth unit.
[image]

```python
import plotly.graph_objects as go
surface_colormap=[[1,2],[2,3],[3,4],[4,5],[5,6],[7,8]]
fig= go.Figure(data=go.Surface(
    x=[[1,1],[2,2],[3,3],[3,3],[2,2],[1,1]],
    y=[[1,1],[2,2],[3,3],[3,3],[2,2],[1,1]],
    z=[[2,1],[2,1],[2,1],[1,0],[1,0],[1,0]],
    surfacecolor=surface_colormap,
    colorscale ='Jet', 
    hovertext =surface_colormap
    ))

fig.show()
```
- use Mesh3D
use i, j, k indices to conncet three points as a triangle
[image]

```python
fig= go.Figure(data=go.Mesh3d(
    x=[1,2,3,1,2,3],
    y=[1,2,3,1,2,3],
    z=[1,1,1,2,2,2],
    intensity=[1,2,3,1,2,3],
    i=[0,1,1,4],
    j=[1,3,2,2],
    k=[3,4,4,5],
    colorscale ='Jet',
    ))

fig.show()
```

ldr426, 2022-08-04:
To just get a scatter  (data point cloud):

``` python
import plotly.graph_objects as go

lon=MVBS_ds.longitude.values[:]
lat=MVBS_ds.latitude.values[:]
depth=MVBS_ds.echo_range.values[:]
plotly_data=[]

for depth in MVBS_ds_spec.echo_range.values:
    plotly_data.append(go.Scatter3d(x=lon, y=lat, z=np.full(len(lon),-depth),
                             mode='markers',
                             marker=dict(
                                 color=MVBS_ds.sel(echo_range=depth).Sv.values,
                                 colorscale='Jet',
                                 opacity=0.5)))

fig = go.Figure(data=data)
fig.show()
```

leewujung, 2025-05-01:
@Galaxy3696 : it would be interesting to see how `plotly` compares with `pyvista` in terms performance to plot 3D curtains.

Also, you can do the initial test using the dataset, but it is very reduced (small) and not realistic. If you can also try a larger dataset using those MVBS zarr files @sashalai64 is working on, that will give a better indication of performance.

Galaxy3696, 2025-05-19:
[image]
The above chart was created using Plotly. I would like to confirm if there are any areas that could be improved. Also, should this implementation directly replace the previous one, or should the previous version be retained?

leewujung, 2025-06-14:
@Galaxy3696 : Thanks for the testing! Plotly output looks nicer than Pyvista. 

Sorry I missed your question here. I haven't merged your code to `main` because you replaced the pyvista functionality completely, but I think it would indeed be better to retain the previous version, and make it an option for users to choose either pyvista or plotly.

Could you please make this change to #192? Thanks!