echostack-org/echoshader #55: DatetimeRangeSlider is not accessible currently

ldr426, 2022-08-14:
Currently datetimeRangeSlider doestn't exist in panel ( v0.13.1, 2022-7-30 )
AttributeError: module 'panel.widgets' has no attribute 'DatetimeRangeSlider'
See this widget in :
https://panel.holoviz.org/reference/widgets/DatetimeRangeSlider.html#widgets-gallery-datetimerangeslider'
When applicable, use below code to get a panel widget : 
```
self.datetime_range_input = panel.widgets.DatetimeRangeSlider(
        name='Datetime Range Input',
        start=self.lower_time,
        end=self.upper_time,
        value=(self.lower_time, self.upper_time))
```

leewujung, 2023-05-20:
@ldr426 : could you check if this is now resolved in new Panel versions?

ldr426, 2023-08-10:
Solved in new panel version

closed 2023-08-10 by ldr426