echostack-org/echoshader #59: Heat map for bird's-eye view

ldr426, 2022-09-08:
Currently, we can use heatmap to get a bird's-eye view. But sometimes, results could be weird.
``` python
def plot_birds_eye(MVBS_ds, surface = 0, cmap = 'jet'):
    '''
    Plot a bird's-eye view for Sv value

    Parameters
    ----------
    MVBS_ds : xarray.Dataset
        MVBS Dataset with coordinates 'ping_time', 'channel', 'echo_range'
        MVBS Dataset with a specific frequncy
        MVBS Dataset has been combined with longitude & latitude coordinates using echopype
        
    surface : int, default : 0
        Determines which depth surface is seen
        
    cmp : str, default: 'jet'
        Colormap

    Returns
    -------
    hvplot.heatmap
        See more in:
        https://hvplot.holoviz.org/reference/pandas/heatmap.html#pandas-gallery-heatmap
        
    Examples
    --------
        MVBS_ds_ = MVBS_ds.sel(channel='GPT  18 kHz 009072058c8d 1-1 ES18-11')
        
        plot_birds_eye(MVBS_ds_, surface = 20)
        
    '''
    date_time = pandas.to_datetime(MVBS_ds.ping_time.values)

    day_time =  numpy.array(numpy.array(MVBS_ds.ping_time.values,dtype='datetime64[ms]')- \
                            numpy.array(MVBS_ds.ping_time.values,dtype='datetime64[D]'), 
                            dtype= 'datetime64[ms]')

    day_time = pandas.to_datetime(day_time)

    Sv = MVBS_ds.sel(echo_range = surface).Sv.values

    data = {'date_time': date_time.day.astype(int),
            'day_time': day_time,
            'Sv':Sv}
    
    df = pandas.DataFrame(data)
    
    return df.hvplot.heatmap(x='day_time', 
                             y='date_time', 
                             C='Sv', 
                             cmap=cmap, 
                             xlabel='Date',
                             ylabel='Time in A Day',
                             clabel='Sv (dB)',
                             title='From '+ str(date_time[0]) + ' to '+str(date_time[-1])).aggregate(function=numpy.mean)
```

leewujung, 2023-05-20:
Not sure what this means. @ldr426 : since you've revamped the notebook, can are elaborate what "weird" means here (modify your initial comment above)?

ldr426, 2023-08-10:
This snippet does not work now. Reserve as a reference.