echostack-org/echoshader #70: Create dedicated notebooks/scripts for generating organized dataset

leewujung, 2023-04-10:
This follows from #3:

In 2022, @ldr426 created notebooks under [./docs/source/gsoc/2022](https://github.com/OSOceanAcoustics/echoshader/tree/main/docs/source/gsoc/2022) that contain data pulling components as part of showing viz functionalities.

This year let's create dedicated notebooks or scripts for pulling large volumes of data and keep them in s3 bucket for developing viz access/plotting/optimization.

A few things to note:
- Current notebooks create MVBS from each .raw files separately and use `xr.open_mfdataset` to combine them for viz. The final version of notebooks/scripts this year should combine multiple `EchoData` objects _before_ calibration and computing MVBS to generate organized data.
- The above hinges on 2 critical echopype updates @lsetiawan will be working on in the next couple months:
  - overhauling `combine_echodata` (which is for combining the `EchoData` objects efficiency as zarr stores)
  - optimize efficiency in `compute_MVBS`

ldr426, 2023-05-29:
| Category | Resolution  |
|---|---|
|OOI 2017 Aug|[1 day with 1 min along ping_time, 1 m for depth (OOI 2017 Aug)](https://drive.google.com/drive/folders/19_Wd1ugIsrMBjJbi8tnHdq-E55LpRMty?usp=sharing)|
|Hake Survey|[1 day with 10 min along ping_time, 5 m for depth (Hake Survey)](https://drive.google.com/drive/folders/116WhBXQRJFT1jzmZhD_T6RaP5jKRMo35?usp=sharing)|
|OOI 2017 Aug|[30 days with 30 mins along ping_time, 1 m for depth (OOI 2017 Aug)](https://drive.google.com/drive/folders/1caGwWmeMOjgi51R9fMKi7XcLx0tAeZOz?usp=sharing)|
|Hake Survey|[10 days with 30 mins along ping_time, 5 m for depth (Hake Survey)](https://drive.google.com/drive/folders/1t53cGzFNLh16bAgzdoAHH2tIk3Gjo9ma?usp=sharing)|
|OOI 2017 Aug|[2 hr in raw resolution (with 1 sec along ping_time, 0.2 m for depth) (OOI 2017 Aug)](https://drive.google.com/drive/folders/1a5Euv5NdRcKqwh8l8h_YuGKqz0_Zxjf6?usp=sharing)|
|Hake Survey|[2 hr in raw resolution (with 2 secs along ping_time, 0.2 m for depth) (hake survey)](https://drive.google.com/drive/folders/1Xem6ugZGfIS54Ov_V9diblaqBZOb0JAr?usp=sharing)|

ldr426, 2023-05-31:
Data got on EBS now:
<p>Moored (30 days, 2017-08-01 to 2017-08-31)</p>

Range meter bin (m) | Ping time bin (s)
-- | --
1 | 1800
1 | 600
1 | 300
1 | 60
0.5 | 600
0.5 | 300
0.5 | 60
0.2 | 300
0.2 | 60
0.2 |  1 (08/01 - 08/20, no concatenated data generated)

<p>Ship (10 days, 2017-07-24 to 2017-08-02)</p>

Range meter bin (m) | Ping time bin (s)
-- | --
5 | 1800
5 | 600
1 | 600
1 | 300
1 | 60
0.2 | 300
0.2 | 60
0.2 | 2 (no concatenated data generated)

ldr426, 2023-05-31:
Data generating examples (will add annotations and comments later):
[Moored_Echosounder_Data_Generate.ipynb ](https://drive.google.com/file/d/1TBLBO_G2HkmRG-EQnQ-wP5IiMzJvAOKS/view?usp=share_link)
[Pacific_Hake_Survey_Data_Generate.ipynb](https://drive.google.com/file/d/1LAlVBe1ggM2uBZkha5knPWKBPbfD7m01/view)

leewujung, 2023-06-26:
We can close this now that #42 and #43 are both closed. See #92 for summary and links.

closed 2023-06-26 by leewujung