echostack-org/echoshader #75: Test all 2022 echoshader features

leewujung, 2023-05-21:
The first step for getting back to the development is to revamp and test all 2022 echoshader features.

@ldr426 : please list in comments below what you have tested and document what works and what doesn't (and what the problem or suspected problem is if it does not work).

ldr426, 2023-05-31:
# Problem 2
When using `invert_yaxis = True`, echogram does not show immediately.
You can see the gram after scrolling down along the y-axis. But the value on the axis is not correct.
[image]
[image]

# Possible Cause
Unknow. The old version does not have this problem.

# Temporary Solution
Do not use option `invert_yaxis`. There must be approach to invert y-axis correctly, which needs to be dived into later.
[image]

ldr426, 2023-05-31:
# Problem 3
There is no `box select` in tool bar.
[image]

# Possible Cause
Unknown. In the old version, the tool icon appears in default.

# Solution
[Explicitly claim this tool in options](https://holoviews.org/user_guide/Linked_Brushing.html#javascript-based-linked-brushing)
[image]
[image]

ldr426, 2023-05-31:
# Problem 4
Errors below appear when running echogram.
[image]

# Possible Cause
May come from `rasterize`
[image]

# Solution
No impact on running. Not emergent. Need to be checked later.

ldr426, 2023-05-31:
# Problem 5
Errors below appear when tunning control widgets on panel.
[image]
# Possible Caues
May come from conflict bewtwen different controlled  grams.
One control panel is responsible for multiple grams.

# Solution
No impact much on running. Not emergent. Need to be checked later.

ldr426, 2023-05-31:
# Problem 6
No color map rendering when plotting 3D curtain.
[image]

# Possible Caues
Unknown.

# Solution

Problem 5
Errors below appear when tunning control widgets on panel.
image

Possible Caues
May come from conflict bewtwen different controlled grams.
One control panel is responsible for multiple grams.

Solution
No impact much on running. Not emergent. Need to be checked later.

ldr426, 2023-05-31:
# Functions Tested List
## Simple Hist
[image]

## Echogram for single gram
[image]

## Box Select
[image]

## Echogram for multiple grams
[image]

## Statistic Table in Echostats
[image]
[image]

## Statistic histogram(single, overlay, layout) in Echostats
[image]

[image]

[image]

## Map in Echomap
[image]

## Curtain in Echomap
[image]

ldr426, 2023-06-02:
# Problem 7 (Solved)
No `hover` tool when plotting overlay gram+box.
[image]

# Possible Caues
 Unkown

# Soultion
Explicitly declare `hover` tool in holoviews opt.
[image]
[image]

ldr426, 2023-06-02:
# Problem 8
No pyvista shown in EC2

# Cause
[ImportError: libGL](https://stackoverflow.com/questions/55313610/importerror-libgl-so-1-cannot-open-shared-object-file-no-such-file-or-directo)
or
https://github.com/pyvista/pyvista/issues/177

# Solution
in Ubuntu 
`sudo apt install libgl1-mesa-glx`
When running jupyter:
``` python
import pyvista
pyvista.start_xvfb()
```
or
https://docs.pyvista.org/version/stable/getting-started/installation.html#running-on-remote-servers

Hoxbro, 2023-06-07:
Problem 1: is caused by an upstream change in Jupyter, ref: https://github.com/holoviz/holoviews/issues/5699#issuecomment-1569025301. We have tried to reduce this by setting `inline=False` in 1.16.1. 

Problem 4: It seems like a bug in Holoviews. If possible, please report it with a MRE. 

Problem 9: This was a bug in Panel and is fixed in the latest release (1.1).

leewujung, 2023-06-26:
Great @ldr426 that you have a list of all the problems and causes and solutions for a subset of them in #92. I think we can close this issue now. I'm putting the "good future ref" label so that we can come back to them once the refactoring is done.

closed 2023-06-26 by leewujung