echostack-org/echoshader #76: Set up Panel on a cloud instance

leewujung, 2023-05-21:
@ldr426 : please put in the detailed steps here (and remove this line 🙂)

ldr426, 2023-05-31:
https://www.notion.so/EC2-faeceebfc16f45509d3f4c18c25f2c22?pvs=4

ldr426, 2023-06-02:
Using `.servable()` to declare a server, example:  https://echoshader.readthedocs.io/en/latest/gsoc/2022/Benchmarking.html#admin-panel

Input below command in conda terminal: 
`panel serve app.py --address 0.0.0.0 --port 5006 --allow-websocket-origin="*"`

Configure secure group(Customize tcp and port permission) on EC2.

ldr426, 2023-06-02:
Use `nohop cmd &` to run process even after closing terminal.
Use `top` to check and kill processes.

leewujung, 2023-06-26:
Add this link from Panel for later reference: https://panel.holoviz.org/how_to/server/programmatic.html

With a few testing examples running, I think we can close this issue now.

closed 2023-06-26 by leewujung