echostack-org/echoshader #77: Documenting existing echoshader structure

leewujung, 2023-05-21:
To facilitate refactoring the echoshader package, we need a clean documentation of what the current echoshader structure and interaction look like. Please follow the UML diagram conventions and put a few references for the diagrams in the comments below.

### TODOs
- [ ] Class diagram and interaction diagram for methods
- [ ] Definitions of data input/output format for the classes and methods

ldr426, 2023-05-24:
[Class diagram and interaction diagram for methods](https://drive.google.com/file/d/1eW6uJJre5nuUUSPq8ux0xlmHdzI6ae75/view?usp=sharing)

ldr426, 2023-05-25:
[Definitions of data input/output format for the classes and methods](https://www.notion.so/Echoshader-Definitions-808e542d92884aaab46edb9969f5eb33?pvs=4)

leewujung, 2023-06-11:
This issue can be closed now!

closed 2023-06-11 by leewujung