echostack-org/echoshader #78: Benchmarking different sizes of datasets

leewujung, 2023-05-22:


ldr426, 2023-06-12:
# Testing env
Instance | vCPU* | CPU Credits / hour | Mem (GiB) | Storage | Network Performance
-- | -- | -- | -- | -- | --
t2.xlarge | 4 | 54 | 16 | EBS-Only | Moderate

# Echogram and EchoStats
For dataset `Moored_range=0.2_ping=60s_start=2017-08-01-00-00_end=2017-08-31-23-59`
Best resolution for **OOI** concatenated data:
CPU times: user 54.6 s, sys: 21.3 s, total: 1min 15s, Wall time: 1min 17s
[image]

For dataset `Hake_range=0.2_ping=60s_start=2017-07-24-00-00_end=2017-08-02-23-59`
Best resolution for **Hake** concatenated data:
Kernel died when running. Can render echogram but fail to do statistic stuff.

For dataset `Hake_range=0.2_ping=300s_start=2017-07-24-00-00_end=2017-08-02-23-59`
Second best resolution for **Hake** concatenated data:
CPU times: user 21.9 s, sys: 8.92 s, total: 30.8 s
Wall time: 33.4 s
[image]

# Echogram and EchoMap
For dataset `Hake_range=0.2_ping=60s_start=2017-07-24-00-00_end=2017-08-02-23-59`
Best resolution for **Hake** concatenated data:
CPU times: user 69.9 ms, sys: 11.3 ms, total: 81.2 ms
Wall time: 79.9 ms
[image]

**Cannot render striped curtain.**
Curtain can be displayed with `Hake_range=1_ping=300s_start=2017-07-24-00-00_end=2017-08-02-23-59`:
[image]

valentina-s, 2023-10-13:
This is done.

closed 2023-10-13 by valentina-s