echostack-org/echoshader #80: Highlight Ship Track Section and Corresponding Echogram

ldr426, 2023-06-02:
Fisheries scientists often need to analyze the fish and zooplankton composition at specific geographic locations. Currently, echoshader visualizes ship tracks on a map based on gram range, but I find it challenging to understand the corresponding echograms for a particular section of the track (reverse operation). Therefore, I would like to propose a new feature that allows users to highlight a section of the ship track and view the corresponding echogram data.

User should be able to select a specific section of a ship track on the map.
Upon selecting the section, the corresponding echogram data should be displayed.

leewujung, 2023-06-30:
Hey @ldr426 : How about breaking this issue up into 3 different issues? since they are related but each involves different work. You can hover mouse over each of the bullet points that click the upper-right corner to turn each into an issue, and this current one will be come the "tracking issue," which is nice. :)

valentina-s, 2023-10-13:
All these are refactored in 0.1.0.

closed 2023-10-13 by valentina-s