echostack-org/echoshader #81: Plot Histogram for Non-Rectangular Region

ldr426, 2023-06-02:
Fisheries scientists collaborate with colleagues to verify species identifications in different regions. However, the current functionality of the system does not allow me to plot a histogram specifically for non-rectangular polygon regions. Therefore, I would like to propose a new feature that enables users to plot a histogram based on a non-rectangular polygon region, allowing to analyze the distribution of "hake" in these areas accurately.

Users should be able to select a non-rectangular polygon region.

Relavant reference: [lasso select in holoviews](https://holoviews.org/user_guide/Linked_Brushing.html#box-select-vs-lasso-select)

ldr426, 2023-06-14:
Warning appears when plotting real-time bounds from box select in VS Code:
See: https://github.com/holoviz/holoviews/issues/5731

leewujung, 2023-06-30:
Hey @ldr426 : What do you think if you split this issue into 2: one focuses on using a rectangular area (a box) and the other focuses on using a polygon area, with each of them having the (a) selection and (b) data export component? This way in the current Refactoring milestone you could just focus on the box one, and push the polygon one to the next milestone?

valentina-s, 2023-10-13:
This is issue was split and the remaining parts about polygon select are separate issues now #118, #124. So closing!

closed 2023-10-13 by valentina-s