echostack-org/echoshader #82: Plot Tricolor(Overlay) Echogram for Distinguishing Fish Species

ldr426, 2023-06-02:
Fisheries scientists often work with echogram data to analyze fish species composition. To improve the accuracy of species identification, I would like to propose a new feature that allows users to plot a "tricolor" echogram. This overlay would map three different frequencies to RGB colors, providing a visual representation that helps distinguish between different fish species.

Echoshader should provide an option to generate a tricolor echogram overlay by mapping the selected frequencies to RGB colors.
The resulting overlay should be displayed on the echogram, with different colors representing different frequency bands.

ldr426, 2023-06-09:
X-coordinate cannot show correctly in RGB
[image]

leewujung, 2023-06-30:
The core of this issue (the 3 functionalities) are addressed in PR #93. Additional refinement are recorded in new issues, so I'll close this one now.

closed 2023-06-30 by leewujung