echostack-org/echoshader #94: Refactor Old Version Histogram; Create Histogram for Non-Rectangular Region

ldr426, 2023-06-13:
- (**Refactor**) As a fisheries scientist, John wants to highlight a part of an echogram as he scrolls through to check the Sv distribution (histogram), so that he can distinguish what type of fish were observed.

- (**Refactor**) As a fisheries scientist, John wants to compare the Sv distribution from multiple echosounder channels (frequencies), so that he can determine the likely composition of fish aggregations.

- (**To do**) As a fisheries scientist, John wants to build histogram from a polygon selection, so that he can analyze the Sv distribution of specific areas within an echogram and identify the types of fish present in those areas.

Refactor: https://github.com/OSOceanAcoustics/echoshader/issues/20
Users should be able to select a non-rectangular polygon region to create hists and statistic tables.

leewujung, 2023-06-30:
Will also be a good idea to separate out these "user stories" into different issues, especially between the box and the polygon cases.

valentina-s, 2023-10-13:
Histograms are refactored in #146. Polygon selection is separated in its own issue: #114. Closing!

closed 2023-10-13 by valentina-s