echostack-org/echoshader #97: Plots update slowly when updating Sv_range frequently

ldr426, 2023-06-15:
After changing the values several times for the Sv_range the demo stopped making any changes: most probably it queued all the options uesers give it. That made click even more which most probably made it worse

We could register only the last one. If there is a way to cancel or time out a sequence of requests it will make it better.

_Originally posted by @valentina-s in https://github.com/OSOceanAcoustics/echoshader/pull/93#pullrequestreview-1480614013_

leewujung, 2023-06-30:
Let's push this to after the Refactoring milestone is done and we can focus on efficiency and timeout issues, since this is likely going to be a recurrent thing for other viz types too.

LOCEANlloydizard, 2026-10-30:
with the app' introduced in #278 and refined in #283, we provide configurable display limits through max_pings and max_range_samples. closing this issue against the old implementation: if Sv limit changes cause delays in the current app, we can track that with a new reproducible example!

closed 2026-10-30 by LOCEANlloydizard