echostack-org/echoshader #98: Do the overall change and check on `MVBS_ds`

ldr426, 2023-06-15:
We can do the overall change from `MVBS_ds` to just `ds`. 

To ensure people put in the right type of dataset, we should add a input checking method that is called in `Echogram.__init__` to check for the following:
- existence of the `Sv` data variable
- `Sv` data variable should have dimensions ("channel", "ping_time", "echo_range")

We should also allow `depth` in the last dimension, but we can do that later -- I'll create an issue so that we do not forget.

_Originally posted by @leewujung in https://github.com/OSOceanAcoustics/echoshader/pull/93#discussion_r1231323523_

leewujung, 2025-06-14:
I think this is addressed by @Galaxy3696 as part of #192 .

@shivam096 : Could you make the suggested changes here and also make sure it works with the latest MVBS dataset from the ship? (I'll point you to some offline.)

LOCEANlloydizard, 2026-10-30:
this is I think mostly addressed by the current [dataset validation](https://github.com/echostack-org/echoshader/blob/main/echoshader/validation.py) called during accessor initialization in core.py: it checks for Sv, dimensions (channel, ping_time, vertical), and nonempty one-dimensional coordinates. Supported vertical dimensions are range_sample, echo_range, and depth. It also checks channel uniqueness and valid, strictly increasing datetime ping times. The notebooks in #284 demonstrate native Sv, MVBS, and resampled geometries!

closed 2026-10-30 by LOCEANlloydizard