echostack-org/echoshader #106: Allow passing custom plotting options to echogram

leewujung, 2023-06-24:
Currently the plot options such as `width` and `height` are fixed as they are initiated in init as `self.gram_opts`. It would be awesome if we could preserve the way holoviews plots are customized and enable doing something like:
```python
ds_MVBS.eshader.echogram_multiple_frequency(layout="composite").opts(
    "width": 1200, "height": 400, ...
)
```

ldr426, 2023-07-05:
Please directly change the property of Echogram class:
```python
self.gram_opts = {
    "Image": {
        "cmap": "jet",
        "colorbar": True,
        "tools": ["box_select", "lasso_select", "hover"],
        "invert_yaxis": False,
        "width": 600,
        },

    "RGB": {"tools": ["hover"], 
            "invert_yaxis": False, 
            "width": 600
            },
}
```

leewujung, 2023-07-05:
I did this on the cruise just so that I can change the sizes, but the users should not have to touch the underlying code. We need to support the use case as I listed above.

valentina-s, 2023-10-13:
This is now implemented in version 0.1.0.

closed 2023-10-13 by valentina-s