echostack-org/echoshader #109: An environment.yaml?

leewujung, 2023-06-27:
Echoshader has some large dependencies, as shown in @ldr426 's notes [here](https://frosted-hamster-458.notion.site/EC2-set-up-faeceebfc16f45509d3f4c18c25f2c22). It will be great to have an environment.yaml so one can easily install things.

From the notion notes, some packages are installed from specific channels outside of `conda-forge`, and it seems that we can specify the channel for specific packages using syntax like [this](https://github.com/conda/conda/issues/5693#issuecomment-505579189).

LOCEANlloydizard, 2026-10-13:
I'm closing this as I think we no longer need it? pyproject.toml now lists the project dependencies (#250), and Conda/Mamba can still be used to create the Python environment, with Echoshader then installed via pip? @leewujung

closed 2026-10-13 by LOCEANlloydizard

leewujung, 2026-10-13:
Yes, agreed! This is definitely outdated.