echostack-org/echoshader #126: Simplify echogram <--> map/curtain by removing the control direction select

ldr426, 2023-07-13:
discussed yesterday that it would be much more intuitive to use the echogram <--> map/curtain functionality _without_ this menu selector (so that user only needs to worry about the `update_echogram_button` and `update_positions_button`). Could you make a new issue to tackle after this PR is merged?

_Originally posted by @leewujung in https://github.com/OSOceanAcoustics/echoshader/pull/122#discussion_r1262760379_

leewujung, 2023-07-18:
Just to add more detail to this, from https://github.com/OSOceanAcoustics/echoshader/pull/122#pullrequestreview-1528971852:
> * `link_to_track`  and `link_to_echogram` are used in the methods `echogram` and `positions` to link the echogram with the map along with the menu selector `self.link_mode_select` -- I think the logic can be worked out to completely remove this menu selector so that users can just work with buttons `update_echogram_button` and `update_positions_button`

ldr426, 2023-08-10:
Erase `update button` but still reserve directional select to make the logic clear in https://github.com/OSOceanAcoustics/echoshader/pull/131

valentina-s, 2023-10-13:
This now is handled with track/echogram control mode without buttons. Closing for now. It can be revisited if a smoother implementation is found.

closed 2023-10-13 by valentina-s