echostack-org/echoshader #179: Add functionality to connect echogram and ship track

leewujung, 2025-02-05:
Echoshader has the functionality to plot echogram and the associated ship track (lat/lon), but the functionality to connect them back and forth is not currently available. This is something that's very useful for data exploration.

Below is a quick sketch during capstone meeting:

[image]

Galaxy3696, 2025-02-17:
When users interact with the ship's track, they will click a geographic coordinate (longitude/latitude) on the track and retrieve the ping_time. However, echograms should inherently represent a continuous time period of acoustic data, not just an instantaneous snapshot.  So  if we will allow users to define a time window (e.g., ±10 minutes) around the selected `ping_time` to generate the echograms.

leewujung, 2025-02-18:
@Galaxy3696 : Yes, and you could also allow user to drag a box that circles part of the track, and that will give you a range of `ping_time` too.

leewujung, 2025-06-14:
Hey @shivam096 : assigned you to this per our conversation on Thursday as the new mechanism to connect the shiptrack and echogram.

LOCEANlloydizard, 2026-10-30:
Implemented in #278 and refined in #283. Map selection updates the echogram’s time window, and the track highlights the active interval!

closed 2026-10-30 by LOCEANlloydizard