echostack-org/echoshader #180: Add region browser functionality

leewujung, 2025-02-05:
A typical echogram analysis output is echogram "regions" - polygons that mark the boundary of a target animal type, like a fish species. It is very useful to have a "region browser" for users to step through all the regions sequentially or via a drop down menu. This requires pulling the right echogram section (typically based on time) to plot with the specific region. 

Below is a quick sketch from the capstone meeting:

[image]

Galaxy3696, 2025-02-10:
Manually selecting a part of the echogram and dynamically plotting a histogram based on the selected data is straightforward; we don’t need to modify the existing code, as shown in the image below.

[image]
If we want to make a more precise selection,I think providing a time selector may be a good idea.

leewujung, 2025-02-10:
@Galaxy3696 I think you may have misunderstood what the issue is describing. Please read again and let me know if you have questions. @sashalai64 maybe you could discuss together?

sashalai64, 2025-02-10:
Sure, we can discuss together.

ldr426, 2025-02-11:
@Galaxy3696 @sashalai64 If anyone is interested in implementing this, you may try to do some research on the [lasso select in holoviews](https://holoviews.org/user_guide/Linked_Brushing.html#javascript-based-linked-brushing) . The key is how to get the vertices of the selected polygon then get the echo data based off these vertices 

[image]

leewujung, 2025-06-14:
Since the discussions here have diverged from the original intention, I've copied the original title and description to a new issue #195, and changed the current title and description to better fit the discussions in this thread.